echostack-org/echopype #1305: AZFP: Move computing of Sv offset outside of setting of Vendors and test for Sv offset using ASL/Rutgers Data

ctuguinay, 2024-04-17:
* Create the `Sv_offset` dataset in the AZFP parser and pack it in `parser_obj.unpacked_data`
* When ASL/Rutgers AZFP data arrives, write tests for skipped tests from #1304

leewujung, 2024-04-17:
Just realized that the error message can be more specific to add the freq for the second check, so maybe we include that in this next PR also:
```
        # Check if the specified freq-pulse length combination is in the allowable Sv_offset dict
        if pulse_len not in SV_OFFSET[freq]:
            raise ValueError(
                f"Pulse length {pulse_len} us is not in the Sv offset dictionary "
                "for the {freq} Hz channel! "  # add freq info
                "Please contact AZFP Environmental Sciences "
                "and raise an issue in the echopype repository."
            )
```

ctuguinay, 2024-04-17:
Sounds good 👍

leewujung, 2024-11-09:
Hey @ctuguinay - I just noticed this. I think we have the test files now from an email from ASL. Could you add that to the test suite and we can close this? Thanks!

closed 2024-12-30 by leewujung