echostack-org/echopype #1309: Resolve deprecation warnings re: `datatree`, `xr.drop_var`, `numpy.array_equal`

leewujung, 2024-04-24:
There are multiple deprecation warnings that we should address in the next release.

One of the warning is related to the use of `pkg_resources` in `datatree`, but our `datatree` version is pinned to a pretty old version. This one may take some more time.

```
../../miniconda3/envs/echopype_20240415/lib/python3.9/site-packages/datatree/__init__.py:4
  /Users/wujung/miniconda3/envs/echopype_20240415/lib/python3.9/site-packages/datatree/__init__.py:4: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
    from pkg_resources import DistributionNotFound, get_distribution

../../miniconda3/envs/echopype_20240415/lib/python3.9/site-packages/pkg_resources/__init__.py:2832
../../miniconda3/envs/echopype_20240415/lib/python3.9/site-packages/pkg_resources/__init__.py:2832
../../miniconda3/envs/echopype_20240415/lib/python3.9/site-packages/pkg_resources/__init__.py:2832
../../miniconda3/envs/echopype_20240415/lib/python3.9/site-packages/pkg_resources/__init__.py:2832
../../miniconda3/envs/echopype_20240415/lib/python3.9/site-packages/pkg_resources/__init__.py:2832
../../miniconda3/envs/echopype_20240415/lib/python3.9/site-packages/pkg_resources/__init__.py:2832
  /Users/wujung/miniconda3/envs/echopype_20240415/lib/python3.9/site-packages/pkg_resources/__init__.py:2832: DeprecationWarning: Deprecated call to `pkg_resources.declare_namespace('sphinxcontrib')`.
  Implementing implicit namespace packages (as specified in PEP 420) is preferred to `pkg_resources.declare_namespace`. See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages
    declare_namespace(pkg)

echopype/tests/calibrate/test_calibrate_ek80.py::test_ek80_BB_complex_multiplex_NaNs_and_non_NaNs[NYOS2105-D20210525-T213648.raw-target_channel_ping_pattern0]
echopype/tests/calibrate/test_calibrate_ek80.py::test_ek80_BB_complex_multiplex_NaNs_and_non_NaNs[NYOS2105-D20210525-T213648.raw-target_channel_ping_pattern0]
  /Users/wujung/code_git/echopype/echopype/calibrate/cal_params.py:318: DeprecationWarning: dropping variables using `drop` is deprecated; use drop_vars.
    da_param = da_param.drop(plb_var)

echopype/tests/calibrate/test_calibrate_ek80.py::test_ek80_BB_complex_multiplex_NaNs_and_non_NaNs[NYOS2105-D20210525-T213648.raw-target_channel_ping_pattern0]
  /Users/wujung/code_git/echopype/echopype/calibrate/range.py:187: DeprecationWarning: dropping variables using `drop` is deprecated; use drop_vars.
    valid_idx = ~beam["backscatter_r"].isel(beam=0).drop("beam").isnull()

echopype/tests/calibrate/test_calibrate_ek80.py::test_ek80_BB_complex_multiplex_NaNs_and_non_NaNs[NYOS2105-D20210525-T213648.raw-target_channel_ping_pattern0]
  /Users/wujung/miniconda3/envs/echopype_20240415/lib/python3.9/site-packages/dask/array/core.py:1712: FutureWarning: The `numpy.array_equal` function is not implemented by Dask array. You may want to use the da.map_blocks function or something similar to silence this warning. Your code may stop working in a future release.
    warnings.warn(

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
```

ctuguinay, 2024-07-23:
The test `echopype/tests/calibrate/test_calibrate_ek80.py::test_ek80_BB_complex_multiplex_NaNs_and_non_NaNs` also produces:

```
/home/exouser/echopype/echopype/calibrate/ek80_complex.py:34: DeprecationWarning: Conversion of an array with ndim > 0 to a scalar is deprecated, and will error in future. Ensure you extract a single element from your array before performing this operation. (Deprecated NumPy 1.25.)
    L = int(np.round(tau * fs * slope * 2.0))  # Length of hanning window
```

ctuguinay, 2024-07-23:
Also I'm unable to recreate:

```
echopype/tests/calibrate/test_calibrate_ek80.py::test_ek80_BB_complex_multiplex_NaNs_and_non_NaNs[NYOS2105-D20210525-T213648.raw-target_channel_ping_pattern0]
  /Users/wujung/miniconda3/envs/echopype_20240415/lib/python3.9/site-packages/dask/array/core.py:1712: FutureWarning: The `numpy.array_equal` function is not implemented by Dask array. You may want to use the da.map_blocks function or something similar to silence this warning. Your code may stop working in a future release.
    warnings.warn(
```

Looking at this test, it doesn't seem to use `np.array_equal` with any Dask Arrays. I printed out the types of the arrays passed into the `np.array_equal` lines and they were all numpy arrays.

ctuguinay, 2024-07-23:
Also, I'll give the `datatree` unpinning a try tomorrow. We can just put it in `0.9.1` if it's too much to fix.

leewujung, 2024-07-23:
What do you think we push the `datatree` component to later, maybe not even v0.9.1? Since `xarray-datatree` is official now, we should probably do a more thorough investigation before diving in to changes.

ctuguinay, 2024-07-23:
Oh yeah, you're right, let's do that