echostack-org/echopype #1312: Enable End to End Lazy Loaded / Delayed Computation for `compute_Sv`

ctuguinay, 2024-04-24:
From the conversation regarding the merged PR #1302, it's clear that one cannot run a fully delayed `compute_Sv` on EK80 Broadband Complex data (one such example where `.compute()` is called is in `compress_pulse`, and having a fully delayed `compute_Sv` would be very nice.

The following tasks are needed to make this happen:
- [ ] Check the status of EK60 and EK80 CW data calibration: Can they be done fully delayed? If they cannot, identify and fix any forced computation.
- [ ] Look into #1212. This is delayed computation, but memory leaks.
- [ ] Modify `compress_pulse` to not have any `.compute()`. Might be as simple as just not calling it.

I'm leaving this as 1 issue to be done for Milestone 0.9.0, but this can be done in separate PRs and in different milestones if it becomes too much for one PR

leewujung, 2024-04-24:
#1311 addresses the last point. It also includes other components required to handle lazy-loaded `EchoData` object for calibrating broadband complex data.

ctuguinay, 2024-07-03:
Closed by merging of #1311 and #1331

closed 2024-07-03 by ctuguinay