echostack-org/echopype #1346: Provide vessel speed over ground when available

gavinmacaulay, 2024-06-22:
I miss the speed_over_ground variable when reading in Simrad EK data files - is it somewhere? A search of the echopype code didn't show up any code dealing with extracting speed from NMEA data, so perhaps not?

The sonar-netcf4 convention has 'speed_over_ground' as a variable under Platform (v1) or under Platform/NMEA (v2).

leewujung, 2024-11-13:
For `sound_over_ground`, you can try set a breakpoint **after** https://github.com/OSOceanAcoustics/echopype/blob/2813a25437fd9985411c96dc0c3cc8b7d223dc05/echopype/convert/api.py#L472
(`parser.parse_raw()` actually parses the data)
This way you can inspect what is in the parsed data object (it’s a dict of dict of dict structure, so not the easiest to work with -- this came from PyEcholab and we may want to restructure down the line if there’s extra time).

For EK80 data, the NMEA data are parsed into the `parser.nmea` attribute
https://github.com/OSOceanAcoustics/echopype/blob/2813a25437fd9985411c96dc0c3cc8b7d223dc05/echopype/convert/parse_base.py#L64
You can see if this variable is parsed in there in the test files we have.
If not, you may have to trace back to the parser code to see why that is not parsed properly.
It could also be something that Simrad added more recently (the .raw data format still gets updated regularly), and our test files may not include that.

@gavinmacaulay: could you provide a test file (ideally <5 MB, the smaller the better) so that @NikiDab could take a stab at this?

Thanks!

gavinmacaulay, 2024-11-13:
I don't have example files that small. However, a 35MB one is a available here: https://aqualyd-my.sharepoint.com/:i:/g/personal/enquiries_aqualyd_nz/EdZq6IzuN8hNpIEpnh0xhv4BaJLTR0l4nLyxfTbIgSZ0Xw?e=exdtWN. The link will be valid for 1 week, so take a copy if you need.

NikiDab, 2024-11-14:
Thank you. I will download the file.

-Niki

leewujung, 2024-12-27:
Hey @spacetimeengineer : I just learned that @NikiDab will not be working on echopype in the foreseeable future, so I'm reassigning this to you. This issue should be more singular.

NikiDab, 2025-03-05:
Hey @leewujung : 

1) I ran a Python script over the test data Gavin has provided. Please see a screenshot of the Echodata object that is being generated as the output. According to this output, the latitude, and the longitude data are available under "Platform" Data variables but not under "Platform/NMEA".

[image]

2) I ran the code in VS Code debug session and placed break points at the following places as you instructed. As I run the code it appears that both parser.parse_raw() and the self.NMEA are None. 

[image]
[image]

leewujung, 2025-03-05:
@NikiDab : since your output of `open_raw` does contain lat/lon, it does not make sense `self.NMEA` is empty. Maybe check where you set the breakpoints to make sure you actually parsed the data. 

Based on the outcome of  `open_raw`, you will need to add that field into the ek parser.

NikiDab, 2025-03-05:
@leewujung Thank you for your quick response. I edited my comment above to add more clarity to my explanations. Could you look into it and let me know? Thanks, Niki

leewujung, 2025-03-05:
@NikiDab : Following from the above: I wonder if you have evaluated `parse_raw()` before examining the values in `self.NMEA`?

gavinmacaulay, 2025-03-05:
@NikiDab : Once you get to parsing the NMEA sentences, the two that contain speed over ground are the VTG and RMC types. My example dataset probably only has RMC, but you're bound to find VTG messages in other .raw examples files that echopype has. Note that there are two formats for the VTG message. Details on the VTG and RMC messages can be found [here](https://gpsd.gitlab.io/gpsd/NMEA.html).

NikiDab, 2025-03-10:
Hi @leewujung, here is an update:

1) Yes, parser.nmea does contain data and it is not empty. It is a dictionary with two keys, "timestamp" and "nmea_string". The value for both keys are lists with a length of 8019 data points.

2) So, the first step is to follow and understand how the lat/long data are parsed from the nmea sentences, then to generalize it to parsing speed_over_ground. 

3) Before diving deeper into the code, I was curious if one can find a relation between the ed['Platform']['latitude'] data with the NMEA sentences in the parser.nmea. One observation is that: ed['Platform']['latitude'] contains 440 data points, whereas parser.nmea['nmea_string'] has 8019 data points. Also, the first data point in the ed['Platform']['latitude'] is -36.63854278, while in parser.nmea['nmea_string'] is $SDVLW,5128.374,N,5128.374,N. At this stage, I am not sure how these two sets of data are related, but there are gaps in my knowledge. I will investigate the code, and as I gain a better understanding of its workings,I believe the pieces will fall into place to create the full picture.

gavinmacaulay, 2025-03-10:
@NikiDab, there are several types of NMEA strings that end up in the Simrad .raw files, many of which don't contain lat/lon information. This will be why there are 8019 NMEA sentences but fewer extracted lat/lon values. For example, $SDVLW sentences contain the distance that the ship has travelled since some point in the past (5128 n.miles in your example). A little tip is that the first three characters in the NMEA sentence header (e.g., $SD) are mostly irrevelant and can vary ($GP is quite common). It is the last three that mean something (e.g., the RMC and VTG I mentioned in an earlier comment and the VLW you're seeing).

NikiDab, 2025-03-20:
Hi Wu-Jung,

After exploring the Echopype package, I found that in set_groups_base.py, line 180, there is a private method called _extract_NMEA_latlon(), which appears to extract latitude and longitude information. Could you confirm if this is correct? If so, I believe I should implement a similar method to extract speed_over_ground using the same approach. Please let me know your thoughts. Thanks, -Niki

[image]

leewujung, 2025-03-20:
@NikiDab I think if you evaluate the code it'll be clear that when this gets run through.

leewujung, 2026-01-02:
@dbashford-NOAA @gavinmacaulay - Sorry took me a while to get to this!

So echopype does two things right now for NMEA datagrams:
- in `set_nmea` function:  it stores all the datagram in the `Platform/NMEA` group, which contains data variable `NMEA_datagram` (all raw NMEA datagrams) with coordinate `nmea_time` (the corresponding timestamps).
- in `_extract_NMEA_latlon` function: for lat/lon data, it fishes out the NMEA sentences that contains GPS locations (`NMEA_SENTENCE_DEFAULT = ["GGA", "GLL", "RMC"]`), extracts the lat/lon and timestamps, and saves them as variables `latitude` and `longitude` with coordinate `time1` in the `Platform` group.
  - as @dbashford-NOAA said, these `time1` does not include the speed over ground timestamps, so would create the NaN-padding problem
  - here the `time1` naming came from the definition in convention ver.1

I just checked that in convention ver.2 the same `time` coordinate is shared across `latitude`, `longitude`, and `speed_over_ground` under the "platform position subgroup" : https://nbviewer.org/github/ices-publications/SONAR-netCDF4/blob/v2.0/Formatted_docs/crr341.pdf#page=25
This wouldn't work in the scenario here.

What echopype tries to do is to store the raw data in netCDF without modification (as much as possible), and in downstream functions consolidate different data types onto the same grid (ping time x sample along range, `ping_time` x `range_sample` in the Sv dataset). IMHO this gives a clean data processing flow and allows us to map directly to the concepts of data processing levels and data provenance, so that people would know what has happened to the data when it's shared/downloaded. If we interpolate speed over ground to lat/lon or the other way around, it would change the data from the raw form, which can introduce messiness downstream. 
---- As a related note to this, I think it would be useful to have a raw data convention, and a separate processed data convention. 

Therefore, I would argue for having two different timestamps for lat/lon and speed over ground. I don't like the `time1`, `time2`, etc designation from convention ver.1, because the same `timeN` can mean different things for EK60 and EK80 due to the addition of new datagrams in EK80.

@gavinmacaulay : What if we make it `time_X`, with X being location or speed or something else descriptive? Any suggestions are welcome!

gavinmacaulay, 2026-01-05:
Having something similar to `_extract_NMEA_location` for pulling the speed out of the NMEA messages and populating the `speed_ground` attribute with its' own time coordinate would be fine. The idea behind having non-specific time coordinate names was that any particular time coordinate could apply to a disparate set of attributes, as this can depend on what NMEA messages the relevant data come from.

It isn't clear in the convention, but which attributes use `time1` and which use `time2`, etc, isn't prescribed - programs that read the netcdf files are supposed to work out themselves which uses which from the attribute metadata. This means that the actual time dimension names are irrevelant and the `time1`, `time2`, etc, naming is just there to have some consistency and prevent name clashes. One could have a small deviation to the convention and use something like `time_XYZ` and if reading code did the time dimension name lookup it wouldn't matter (but I don't know how many reading codes do that, if any...).

One of the changes in version 2 of the convention was to require all these non-raw platform attributes to be on the same time coordinate to make the data more directly useful (and still keeping the raw data in a NMEA or similar subgroup). The more important attributes were also duplicated in the sonar beam groups, interpolated onto the ping times to make it even simplier to access those data.