echostack-org/echopype #1358: Store and calibrate EC150 echo strength data

leewujung, 2024-07-13:
Right now our parser handles data from EC150 ADCP fine, but we still need to determine how to store the data properly and enable calibrating the data.

Should we store the data 1) as a separate `Beam_group3`, or 2) along with existing `Beam_group1` or `Beam_group2`?

On the calibration side, we need to make sure all the parameters are stored properly.

Right now we are skipping all datagrams generated by EC150 (#1357) just so that we can just handle data from other channels sitting in the same file.

leewujung, 2024-08-26:
@jmjech @spacetimeengineer this is relevant to #995 - could use your inputs also.

jmetteUni, 2026-10-20:
Dear maintainers, I wanted to ask if this is still the state of handling EC150 entries? Or have there been any progress or thoughts how to include them?

leewujung, 2026-10-20:
Hey @jmetteUni - we still have this on our list but have to work on other issues in the next month or so. Would you be willing to contribute? If so, we can start with discussing how to store things! EC150 would be an interesting case for us to try out the "alternative" arrangement of beam groups in #1721.