echostack-org/echopype #1370: ES60 Error: KeyError: 'transceivers'

markseery, 2024-08-02:
### General description of problem

import echopype as ep
from echopype import open_raw
ed = open_raw('./EXAMPLE.raw', sonar_model='EK60')
ed.to_netcdf(save_path='./')

### Echopype version

latest

### How did you install Echopype (e.g., conda or pip)

pip3

### What is your operating system

MacOS

### Minimal code example

import echopype as ep
from echopype import open_raw
ed = open_raw('./EXAMPLE.raw', sonar_model='EK60')
ed.to_netcdf(save_path='./')

### Error message printouts

```shell
Traceback (most recent call last):
  File "/Users/markseery/Documents/Code/Python/GitHub/genos/blueeconomy/unpackraw.py", line 3, in <module>
    ed = open_raw('./EXAMPLE.raw', sonar_model='EK60')
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/markseery/miniconda3/lib/python3.12/site-packages/echopype/utils/prov.py", line 237, in inner
    dataobj = func(*args, **kwargs)
              ^^^^^^^^^^^^^^^^^^^^^
  File "/Users/markseery/miniconda3/lib/python3.12/site-packages/echopype/convert/api.py", line 435, in open_raw
    setgrouper = SONAR_MODELS[sonar_model]["set_groups"](
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/markseery/miniconda3/lib/python3.12/site-packages/echopype/convert/set_groups_ek60.py", line 62, in __init__
    channels = list(self.parser_obj.config_datagram["transceivers"].keys())
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^
KeyError: 'transceivers'
```

### Example data

https://noaa-wcsd-pds.s3.amazonaws.com/index.html#data/raw/Alaska_Knight/EBS11AK/ES60/

### Related existing issues or PRs

None I could see

### Troubleshooting steps

Try different sonar types in python code

ctuguinay, 2024-08-02:
Could you install the latest Echopype Main Branch instead of version 0.8.4 on Pip?

Terminal Command: `pip install git+https://github.com/OSOceanAcoustics/echopype.git@main`

markseery, 2024-08-02:
Same error with main branch. I will have to see if there is a better code example to use than this:

import echopype as ep
from echopype import open_raw
ed = open_raw('./EXAMPLE.raw', sonar_model='EK60')
ed.to_netcdf(save_path='./')

ctuguinay, 2024-08-02:
Ah I just noticed this but are you by any chance using Python 3.12? I see it in the miniconda env screenshot above. We currently don't have CI testing support for Python 3.12 just yet (we currently test for 3.9 3.10 3.11). Could you try with any of those 3?

This suggestion is a bit of a longshot

markseery, 2024-08-02:
Thanks, I will look to create a VENV and see if it works.

markseery, 2024-08-02:
Ok, thanks. On 3.9.19 I was able to get this code to work:

import echopype as ep

s3rawfile = f"https://noaa-wcsd-pds.s3.amazonaws.com/data/raw/Alaska_Knight/EBS11AK/ES60/L0000-D20110528-T143715-ES60.raw"
print(f"Target file: {s3rawfile}")

#try:
ed = ep.open_raw(
    raw_file=s3rawfile,
    sonar_model='ek60',
    use_swap=True,
    #storage_options={'anon': True}
)
ed.to_netcdf(save_path='./')

Once a .nc file has been created, what is the next step to read it and analyze it?

Thanks!!
Mark

ctuguinay, 2024-08-02:
Of course!

closed 2024-08-02 by ctuguinay