echostack-org/echopype #1371: [Data conversion] AttributeError: 'NodePath' object has no attribute '_raw_paths'

HaileySchmidt-NOAA, 2024-08-02:
### General description of problem

I am trying to run the example Jupyter Notebooks but keep running into an error. When running ed = open_raw(...) I get the error "AttributeError: 'NodePath' object has no attribute '_drv'" followed by "AttributeError: 'NodePath' object has no attribute '_raw_paths'". This happens with all three Jupyter notebooks as well as some of my own data that I tried to use.

### Echopype version

echopype v0.8.4

### How did you install Echopype (e.g., conda or pip)

pip

### What is your operating system

Windows

### Minimal code example

Using my files: 
ed = open_raw('AMLR08.ad2cp', sonar_model='AD2CP')

Using the Jupyter Notebook: echopype_tour.ipynb
ed = ep.open_raw(
            f"s3://{s3rawfpath}",
            sonar_model='EK60',
            storage_options={'anon': True}
        )

### Error message printouts

```shell
AttributeError                            Traceback (most recent call last)
File c:\Users\hailey.schmidt\AppData\Local\anaconda3\envs\Gliders\Lib\pathlib.py:555, in PurePath.drive(self)
    554 try:
--> 555     return self._drv
    556 except AttributeError:

AttributeError: 'NodePath' object has no attribute '_drv'

During handling of the above exception, another exception occurred:

AttributeError                            Traceback (most recent call last)
Cell In[7], line 1
----> 1 ed = open_raw('AMLR08.ad2cp', sonar_model='AD2CP')

File c:\Users\hailey.schmidt\AppData\Local\anaconda3\envs\Gliders\Lib\site-packages\echopype\utils\prov.py:237, in add_processing_level.<locals>.wrapper.<locals>.inner(*args, **kwargs)
    235 @functools.wraps(func)
    236 def inner(*args, **kwargs):
--> 237     dataobj = func(*args, **kwargs)
    238     if is_echodata:
    239         ed = dataobj

File c:\Users\hailey.schmidt\AppData\Local\anaconda3\envs\Gliders\Lib\site-packages\echopype\convert\api.py:491, in open_raw(raw_file, sonar_model, xml_path, convert_params, storage_options, use_swap, max_chunk_size)
    487 tree_dict["Vendor_specific"] = setgrouper.set_vendor()
    489 # Create tree and echodata
    490 # TODO: make the creation of tree dynamically generated from yaml
--> 491 tree = DataTree.from_dict(tree_dict, name="root")
    492 echodata = EchoData(source_file=file_chk, xml_path=xml_chk, sonar_model=sonar_model)
    493 echodata._set_tree(tree)

File c:\Users\hailey.schmidt\AppData\Local\anaconda3\envs\Gliders\Lib\site-packages\datatree\datatree.py:360, in DataTree.from_dict(cls, d, name)
    356 if d:
    357     # Populate tree with children determined from data_objects mapping
    358     for path, data in d.items():
    359         # Create and set new node
--> 360         node_name = NodePath(path).name
    361         new_node = cls(name=node_name, data=data)
    362         obj._set_item(
    363             path,
    364             new_node,
    365             allow_overwrite=False,
    366             new_nodes_along_path=True,
    367         )

File c:\Users\hailey.schmidt\AppData\Local\anaconda3\envs\Gliders\Lib\site-packages\datatree\treenode.py:34, in NodePath.__new__(cls, *args)
     31 def __new__(cls, *args: str | "NodePath") -> "NodePath":
     32     obj = super().__new__(cls, *args)
---> 34     if obj.drive:
     35         raise ValueError("NodePaths cannot have drives")
     37     if obj.root not in ["/", ""]:

File c:\Users\hailey.schmidt\AppData\Local\anaconda3\envs\Gliders\Lib\pathlib.py:557, in PurePath.drive(self)
    555     return self._drv
    556 except AttributeError:
--> 557     self._load_parts()
    558     return self._drv

File c:\Users\hailey.schmidt\AppData\Local\anaconda3\envs\Gliders\Lib\pathlib.py:408, in PurePath._load_parts(self)
    407 def _load_parts(self):
--> 408     paths = self._raw_paths
    409     if len(paths) == 0:
    410         path = ''

AttributeError: 'NodePath' object has no attribute '_raw_paths'
```

### Example data

The issue appears with the example data provided in the sample notebooks

### Related existing issues or PRs

I have not seen any related issues

leewujung, 2024-08-02:
@HaileySchmidt-NOAA: what is your datatree version?

HaileySchmidt-NOAA, 2024-08-02:
I am not sure how to check that, could you please advise?

ctuguinay, 2024-08-02:
Ah yes that's the correct datatree version that we intend to have it pinned on. Perhaps related to #1370, are you also using Python 3.12?

ctuguinay, 2024-08-02:
Ah ok this looks exactly like #1279

HaileySchmidt-NOAA, 2024-08-02:
Yes, I was using Python 3.12. I created a new conda environment using Python 3.10.14 and this resolved the issue for me. Thank you for your assistance!

closed 2024-08-02 by HaileySchmidt-NOAA

ctuguinay, 2024-08-02:
Of course!