echostack-org/echopype #1377: `apply_mask` should only take in boolean masks

ctuguinay, 2024-08-22:
After talking to @leewujung, we should modify `apply_mask` to enforce that the masks passed in are boolean masks.

Users can currently pass in masks that are `1`/`0` and `1`/`NaN`, which stand for `True`/`False`; however, more obscure configurations can be passed in, such as `arbitrary float`/`NaN`, with the current way that `apply_mask` handles and applies masks onto Sv/MVBS data, and this makes it a bit confusing for what the user should pass in.

leewujung, 2024-12-27:
@spacetimeengineer - this is another one that's also more singular that I think could be addressed quickly.

spacetimeengineer, 2025-03-03:
@leewujung

When considering echodata masks, I understand the feature and the intent behind it. However, I wanted to ask about a potential edge case where, in the future, someone might use these masks for more complex operations.

Specifically, I’m thinking about scenarios where multiple mask objects exist within a given session. In such cases, a user may need to perform boolean operations, such as unions or intersections, on those masks. I’ve encountered this myself when working with both lines and circular masks—boolean operations became essential for handling them properly.

I agree that having only True and False values makes for a cleaner input/output system. However, I wonder if allowing additional temporary states could be useful for certain 2D operations on xarray or masks that otherwise wouldn’t be possible. The final result should always resolve to True or False, but intermediate states might facilitate calculations.

What are your thoughts on this? I will try to finish this by the end of the week.

leewujung, 2025-03-04:
@spacetimeengineer : We already allow users to put in a list of masks they would like to combine (element-wise AND) in `apply_mask`. For other possible combinations, I think it'll be better to leave it to the users to go through their own processing to come up with clean masks before using this function.

spacetimeengineer, 2025-03-04:
Acknowledged. Thankyou.

leewujung, 2025-05-01:
@shivam096 : I added you to this issue too along with other `apply_mask` issues. @spacetimeengineer has previously put together #1476 but seems like tests are failing. If you could take a look the tests there that'll be super helpful. Thanks!

shivam096, 2025-05-05:
Hi @leewujung,

I have updated the PR with my comments. Also, I would like to check if the users are expected to provide a boolean input mask?  Or can the input mask be of any type?

leewujung, 2025-05-30:
Resolved by #1502.