echostack-org/echopype #1402: `apply_mask` dimension check should be agnostic of order of dimensions

leewujung, 2024-10-29:
Currently the dimension check in `apply_mask` would check for matching dimension length *and* order for `ping_time` and `depth` (or `range_sample`). I've encountered the situation where the Sv or MVBS data being masked and the mask have the same length of the dimension but not the order. I think we can relax the order requirement since that is something that easily get altered depending on prior operations on these datasets.

NikiDab, 2025-05-03:
Hi @leewujung, I am currently working on this issue. I just noticed I have been unassigned from this one. Please let me know which issue you want me to work on. Thanks. -Niki

leewujung, 2025-05-03:
@NikiDab : you've been on this issue for some time and I haven't seen a PR, so I was not sure if you are working on it. If you can push up a PR in the next couple days that will be helpful. Otherwise @shivam096 can take this on.

NikiDab, 2025-05-03:
Yes, I will push a PR soon in the next couple of days. Thanks.

NikiDab, 2025-05-04:
@leewujung I made a PR, but there is an error, and I am working to resolve it. Thanks.

NikiDab, 2025-05-07:
Hi @leewujung, I have copied a small code snippet from the Echopype below on allowed_dims. 

allowed_dims = [
                ("ping_time", "range_sample"),
                ("ping_time", "depth"),
                ("ping_time", "echo_range"),
                ("channel", "ping_time", "range_sample"),
                ("channel", "ping_time", "depth"),
                ("channel", "ping_time", "echo_range"),
            ]

By making the apply_mask function agnostic to the order of dimensions, do we mean that for example the data dimension could have "ping_time", "range_sample" dimensions in it, except that they can be in any order of either "ping_time", "range_sample" or "range_sample", "ping_time"?

NikiDab, 2025-05-07:
@leewujung  Also, would you mind sharing a test dataset with me? I'd like to run my code on it to ensure everything works as expected before submitting a PR. This should help avoid repeated PRs. Thanks. – Niki

leewujung, 2025-05-30:
This is fixed by #1503.

closed 2025-05-30 by leewujung