echostack-org/echopype #1408: Support for python 3.12

leewujung, 2024-11-09:
Currently our tests only run on python 3.9, 3.10, 3.11, and from various previous error reports (https://github.com/OSOceanAcoustics/echopype/issues/1370, https://github.com/OSOceanAcoustics/echopype/issues/1371, https://github.com/OSOceanAcoustics/echopype/issues/1391), looks like we will need to unpin the `datatree` version to do this. Coincidentally (which is probably not a coincidence :P)`datatree` is now officially supported by xarray (https://github.com/pydata/xarray/discussions/9680), so looks like this is a good time for us to resolve these 2 problems together!

Related issue: https://github.com/OSOceanAcoustics/echopype/issues/1405

ctuguinay, 2024-12-26:
This might get somewhat involved because Xarray DataTree is being more strict in its alignment between children and parent nodes, so I'll detail some of the steps that I'll take once I've discovered all alignment mismatches that pose problems in our CI steps.

leewujung, 2024-12-26:
Yeah, given the error messages I suspected that this was because of the datatree upgrade (#1405) and it would become a large task. Let me know if I can help somewhere.

closed 2025-01-01 by ctuguinay