echostack-org/echopype #1415: Issue on `open_raw()`

dhonanhibatullah, 2024-11-26:
I am trying to run a simple code as follows.
```
from echopype import open_raw

ed = open_raw(
     'SD.raw',
    sonar_model = 'EK60'
)
```

But I got the following error:
```
Traceback (most recent call last):
  File "/usr/lib/python3.12/pathlib.py", line 555, in drive
    return self._drv
           ^^^^^^^^^
AttributeError: 'NodePath' object has no attribute '_drv'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dhonan/Workspaces/temp/test.py", line 3, in <module>
    ed = open_raw(
         ^^^^^^^^^
  File "/home/dhonan/Workspaces/temp/.venv/lib/python3.12/site-packages/echopype/utils/prov.py", line 237, in inner
    dataobj = func(*args, **kwargs)
              ^^^^^^^^^^^^^^^^^^^^^
  File "/home/dhonan/Workspaces/temp/.venv/lib/python3.12/site-packages/echopype/convert/api.py", line 540, in open_raw
    tree = DataTree.from_dict(tree_dict, name="root")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dhonan/Workspaces/temp/.venv/lib/python3.12/site-packages/datatree/datatree.py", line 360, in from_dict
    node_name = NodePath(path).name
                ^^^^^^^^^^^^^^
  File "/home/dhonan/Workspaces/temp/.venv/lib/python3.12/site-packages/datatree/treenode.py", line 34, in __new__
    if obj.drive:
       ^^^^^^^^^
  File "/usr/lib/python3.12/pathlib.py", line 557, in drive
    self._load_parts()
  File "/usr/lib/python3.12/pathlib.py", line 408, in _load_parts
    paths = self._raw_paths
            ^^^^^^^^^^^^^^^
AttributeError: 'NodePath' object has no attribute '_raw_paths'
```

Please note that I am using Python virtual environment, with echopype==0.9.0. Any idea on fixing this?

leewujung, 2024-11-26:
@dhonanhibatullah : I think this is likely due to the datatree version, since we pinned it to an earlier version to avoid conflict. What are the versions of xarray and datatree in your env?

dhonanhibatullah, 2024-11-27:
```
xarray==2024.11.0
xarray-datatree==0.0.6
```

peter-urban, 2024-12-17:
I just ran into the same issue and I think this a duplicate of https://github.com/OSOceanAcoustics/echopype/issues/1391 The solution should thus be to use python 3.11.

dhonanhibatullah, 2024-12-18:
Thanks for the information. I'll close this issue.

closed 2024-12-18 by dhonanhibatullah