echostack-org/echopype #1418: Maximum python-version number for published echopype packages

peter-urban, 2024-12-17:
Echopype provides conda-forge and pypi packages for python 3.12 (and probably even 3.13 and 3.14) even though it is not compatible (https://github.com/OSOceanAcoustics/echopype/issues/1408). This creates difficult-to-understand problems (e.g. https://github.com/OSOceanAcoustics/echopype/issues/1415). 

Given that echopype has such an extensive ci-test-suit that would easily catch such issues, the maximum version for published packages on pypi and conda should be set to the maximum python-version used in the ci-test suit. I am not familiar with build/publish procedure applied, but is it possible to specify a maximum (and also minimum) python version for the package? (e.g. the requires-python field in [project] section of the pyproject.toml (https://peps.python.org/pep-0621/)

ctuguinay, 2024-12-26:
Oh yeah let me try specifying minimum python version in `pyproject.toml`. I'll also work on supporting Python 3.12 these next few days.

leewujung, 2024-12-26:
Our current test matrix only goes up to 3.11, so the errors are not caught. We know that it is likely the datatree upgrade of xarray that triggered these conflicts, which makes this a bigger task (see #1405 and #1408). @ctuguinay thanks for taking this on!

leewujung, 2024-12-27:
Related to this - I just realized that the batch of latest changes in the installation doc page is not in the `stable` RTD version folks see, but in `latest` since we have not released for a while. In [that page](https://echopype.readthedocs.io/en/latest/installation.html) we have a warning about 3.12 not being supported. Since it's been so long, I think we can just wait for #1419 and then release.

closed 2025-01-01 by ctuguinay