echostack-org/echopype #1421: Migrate to use `pyproject.toml` fully

leewujung, 2024-12-29:
Right now the project specification is a confusing combination between `pyproject.toml` and `setup.cfg`, plus `requirements.txt`. It'll be great to migrate to using `pyproject.toml` fully. 

Some resources from a quick look:
- dynamic versioning https://github.com/pypa/setuptools-scm
- https://setuptools.pypa.io/en/latest/userguide/dependency_management.html#python-requirement
- https://github.com/pydata/xarray/blob/main/pyproject.toml
- https://github.com/scipy/scipy/blob/main/pyproject.toml
- https://docs.xarray.dev/en/stable/contributing.html#creating-a-python-environment

oftfrfbf, 2025-02-10:
Not sure if this is within the scope of this issues' intent but just want to give a shout out to the "uv" package and project manager as a possible solution.

[https://docs.astral.sh/uv/](https://docs.astral.sh/uv/)

Just as a comparison to the more traditional "conda" and "pip" managers I am enjoying using 'uv' so far for personal projects — support was just added with Pycharm's latest updates. The biggest feature is fast package management (it's written in Rust) so it has shaved a lot of time off my github action run time. It also starts to consolidate most all of the random python fluff back into the pyproject.toml file, so I can define dependencies, dev-dependencies, and test-dependencies as needed in one central file. It can handle the virutal environment. It can handle building/packaging, testing, and test-matrices. There is now a ".lock" file associated with package installation (similar to what NodeJS has), so package specifications are more structured and you aren't accidentally using packages without properly adding them to the project. There are a ton of other features I'm still exploring but I really like it.

All that said, it would be a bit of a lift to make the changes mentioned above.

If the intent is to just move requirements.txt and requirements-dev.txt to the pyproject.toml file I can help with that — just let me know.

leewujung, 2025-02-10:
@oftfrfbf : uv does look very interesting! But you are right that we may not be able to do this right now... Maybe we can take it with stages -  The first step getting everything into pyproject.toml (this issue), and at a later time support uv?

One thing I haven't looked up on is how to make it such that people can create a regular or dev environment using conda based on pyproject.toml, since we need to include that in the docs. A quick search landed on this: https://github.com/conda/conda/issues/10633 and it's funny that the last comment pointed to `uv`. I saw another thread ended with that too. Any insights here?

Would you be willing to write an issue for using `uv` so that we can re-consider at some point?

oftfrfbf, 2025-02-11:
Thanks, I will add a new issue for uv sometime in the near future.

Not sure how successful I will be given the comments in those github issues but I will create a new fork tonight and try and get the pyproject.toml install working for "requirements.txt" and "requirements-dev.txt." I would probably advocate for the "docs/requirements.txt" to be included in "requirements-dev.txt" unless there is some problem I am not seeing as a constraint?

leewujung, 2025-02-13:
>  I would probably advocate for the "docs/requirements.txt" to be included in "requirements-dev.txt" unless there is some problem I am not seeing as a constraint?

Not really a constraint, but since `docs/requirements.txt` does not live in the project root folder and is only used to create docs, it seems good to not include it in `requirements-dev.txt`?

oftfrfbf, 2025-02-14:
My thoughts were more along the lines of _less-is-more_, if someone is contributing to the documentation then they are doing _development_ for echopype. I'll defer to what you think makes the most sense.

skprasadu, 2025-06-19:
Today, I created a pr for migrating to uv and I do have a pyproject.toml and also we can lock the versions. And when running to PR pipeline, I came across the same issue in the pipeline where, docs/requirements.txt was expected to create document in the pipeline. So I left it there. Other than that there are no requirements.**.txt anywhere and I also updated the documentation.

This might be a big change, but it has its own advantages. Let me know your thoughts on #1509