echostack-org/echopype #1428: Echopype not compatible with zarr 3.0.0 !?

hentatisundberg, 2025-01-16:
Hi, 
It seems like the present version of echopype (0.9.1) is not compatible with the recent release of zarr (3.0.0). I installed echopype with pip in a virtual environment (including the requirements file), and when running open_raw() I get: 

_ImportError: cannot import name 'GroupNotFoundError' from 'zarr.errors' (/Users/jonas/Documents/Programming/python/echopype/venv/lib/python3.13/site-packages/zarr/errors.py)_

Run code and link to example .raw file below. Thanks in advance! 

import echopype as ep
file_path = 'SLUAquaSailor2020V2-Phase0-D20210501-T111709-1.raw'
ed = ep.open_raw(file_path, sonar_model="EK80", use_swap=True)

raw file: 
https://drive.google.com/file/d/1elkyzejWWkceh997VaJhKJD_2FITXii5/view?usp=sharing

leewujung, 2025-01-16:
Hey @hentatisundberg : I suspect this is related to the pending upgrade to python=3.12 (#1405 and #1420 are in the works...). If you install from conda at v0.9.1 it should pull an older version of zarr. But let me know if that's not the case.

hentatisundberg, 2025-01-17:
Yes you are probably right, I should have mentioned I also upgraded to python 3.12. Happy to see this moving forward!

oftfrfbf, 2025-01-19:
For those who are installing using pip installers would it make sense to pin the zarr version temporarily with:
```
zarr>=2,<3
```
as mentioned in the migration guide here:

https://zarr.readthedocs.io/en/latest/user-guide/v3_migration.html

leewujung, 2025-01-19:
Thanks @oftfrfbf - I wonder if we should bump a release with zarr pinned (and potentially netcdf4 too because of #1426).

leewujung, 2025-01-21:
So typically we try to pin as few dependency as possible, unless absolutely necessary, especially when we do < some version. I was hoping we can have the zarr 3.0 advantages "for free," but I guess not! Since the breaking changes would likely require some work/time for us to catch up, it's best to pin it so that folks don't keep on erroring out. 

I'll do another release with zarr and potentially netcdf4 pinned on in the next couple days.

We should try to unpin it as soon as possible, since zarr is one of our key dependencies. This way we don't become outdated and the package slowly becomes unusable since all other packages continue to move forward (kinda like the py>=3.12 case).

leewujung, 2025-01-21:
@hentatisundberg : I changed the issue title so that we can easily see what's needed. I pinned zarr>=2,<3 and will do a quick release v0.9.2 to buy us time to address the changes. Thanks for reporting this!

hentatisundberg, 2025-01-22:
> [@hentatisundberg](https://github.com/hentatisundberg) : I changed the issue title so that we can easily see what's needed. I pinned zarr>=2,<3 and will do a quick release v0.9.2 to buy us time to address the changes. Thanks for reporting this!

Excellent, sounds like a good strategy!