echostack-org/echopype #1432: error importing echopype in fresh installation (version 0.8.4)

marianpena, 2025-01-29:
Hi,
I tried installing echopype in a new environment with conda and I got version 0.8.4. No problems in installation but when I import in Jupyter I get the following error:
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Cell In[3], line 57
      1 # %load_ext autoreload
      2 # %reload_ext autoreload
      3 # %autoreload
   (...)

---> 57 import echopype as ep
     58 import matplotlib as plt
     59 import os, fnmatch

File ~\AppData\Local\Continuum\miniconda3\envs\mimedio2025\Lib\site-packages\echopype\__init__.py:5
      1 from __future__ import absolute_import, division, print_function
      3 from _echopype_version import version as __version__  # noqa
----> 5 from . import calibrate, clean, commongrid, consolidate, mask, utils
      6 from .convert.api import open_raw
      7 from .echodata.api import open_converted

File ~\AppData\Local\Continuum\miniconda3\envs\mimedio2025\Lib\site-packages\echopype\calibrate\__init__.py:1
----> 1 from .api import compute_Sv, compute_TS
      3 __all__ = ["compute_Sv", "compute_TS"]

File ~\AppData\Local\Continuum\miniconda3\envs\mimedio2025\Lib\site-packages\echopype\calibrate\api.py:3
      1 import xarray as xr
----> 3 from ..echodata import EchoData
      4 from ..echodata.simrad import check_input_args_combination
      5 from ..utils.log import _init_logger

File ~\AppData\Local\Continuum\miniconda3\envs\mimedio2025\Lib\site-packages\echopype\echodata\__init__.py:7
      1 """
      2 EchoData is an object that handles interfacing raw converted data.
      3 It is used for calibration and other processing.
      4 """
      6 from . import convention
----> 7 from .echodata import EchoData
      9 __all__ = ["EchoData", "convention"]

File ~\AppData\Local\Continuum\miniconda3\envs\mimedio2025\Lib\site-packages\echopype\echodata\echodata.py:22
     20 from ..utils.prov import add_processing_level
     21 from .convention import sonarnetcdf_1
---> 22 from .widgets.utils import tree_repr
     23 from .widgets.widgets import _load_static_files, get_template
     25 XARRAY_ENGINE_MAP: Dict["FileFormatHint", "EngineHint"] = {
     26     ".nc": "netcdf4",
     27     ".zarr": "zarr",
     28 }

File ~\AppData\Local\Continuum\miniconda3\envs\mimedio2025\Lib\site-packages\echopype\echodata\widgets\utils.py:9
      5 from datatree.render import RenderTree
      7 from ..convention.utils import _get_sonar_groups
----> 9 SONAR_GROUPS = _get_sonar_groups()
     12 def html_repr(value) -> str:
     13     return value._repr_html_()

File ~\AppData\Local\Continuum\miniconda3\envs\mimedio2025\Lib\site-packages\echopype\echodata\convention\utils.py:6, in _get_sonar_groups()
      4 def _get_sonar_groups():
      5     """Utility to reorder convention file by the paths"""
----> 6     group_mapping = sonarnetcdf_1.yaml_dict["groups"]
      7     sonar_groups = {}
      8     for k, v in group_mapping.items():

File ~\AppData\Local\Continuum\miniconda3\envs\mimedio2025\Lib\site-packages\echopype\echodata\convention\conv.py:24, in _Convention.yaml_dict(self)
     21 if self._yaml_dict:  # Data has already been read, return it directly
     22     return self._yaml_dict
---> 24 with resources.open_text(package=convention, resource=f"{self.version}.yml") as fid:
     25     convention_yaml = yaml.load(fid, Loader=yaml.SafeLoader)
     27 self._yaml_dict = convention_yaml

TypeError: open_text() got an unexpected keyword argument 'package'

leewujung, 2025-01-30:
@marianpena - thanks for reporting this! I saw something similar with the test failing in #1429 too, and that was with v0.9.1. Likely dependency version problem. Will try to get this sorted out soon.

leewujung, 2025-02-01:
Looking into this more closely, in #1429 `open_text` triggered warning but not test failures - those were were something. I think this one is an easy fix. But I am curious, @marianpena, what version of `importlib_resources` do you have from this fresh env?

marianpena, 2025-02-04:
This is the version:
Name: importlib_resources
Version: 6.5.2

leewujung, 2025-02-04:
This should be resolved now in #1434, which is now merged to `main` with a few other fixes and pinned zarr and netcdf4 versions -- could you give that a try? (i.e. pull from `main` when installing echopype).

We can't release just yet because the backward compatibility issue with xarray datatree (#1420), but hopefully that could be resolved this week. However, if you don't need to previously converted data, the other parts should work. 🤞

marianpena, 2025-02-05:
Thanks @leewujung. Do you know when this version will be available in conda?

leewujung, 2025-02-05:
I am hoping later this week or early next week (latter is probably more practical... We will need to do whole bunch of testing to make sure we're catching most of the dependency issues.

leewujung, 2025-02-21:
We're releasing v0.10.0 now which covers this. I'll resolve this issue now!

closed 2025-02-21 by leewujung

leewujung, 2025-02-21:
Oops a packaging issue in python 3.12. Will resolve it soon but v0.10.0 won't be today. :/

marianpena, 2025-05-06:
Hi @leewujung, still getting no update from conda. Is it because of the python version?Should I lower it to python 3.10-3.12 or is it coming soon?

leewujung, 2025-05-06:
@marianpena : Please use 3.10-3.12.

marianpena, 2025-05-07:
finally got it, thanks!

leewujung, 2025-05-07:
Awesome!