echostack-org/echopype #1464: [Data conversion] ...Echopype cannot convert Nortek ad2cp

ZHANGGU, 2025-02-19:
### General description of problem

Tried the simplest example according to documentation but it did not work

### Echopype version

echopype==0.9.0

### How did you install Echopype (e.g., conda or pip)

pip install echopype

### What is your operating system

Windows 10 Enterprise

### Minimal code example

ifp = r'c:\adcp\001.ad2cp'
ed = ed.open_raw(ifp, sonar_model='AD2CP')  # for AD2CP file
ed.to_netcdf(save_path=ofp)

Actuall it did run to Echopype code, but stopped when loading python libraries

Problem was due to Numpy version. Have tried upgrade Numpy to 2.1.3 but did not work either

### Error message printouts

```shell
C:\Users\a5484\Documents\Python\zhanggu\Scripts\python.exe C:\Users\a5484\Documents\Python\adcp2nc\convert_ad2cp.py 
<frozen importlib._bootstrap>:488: Warning: Numpy built with MINGW-W64 on Windows 64 bits is experimental, and only available for 
testing. You are advised not to use it for production. 

CRASHES ARE TO BE EXPECTED - PLEASE REPORT THEM TO NUMPY DEVELOPERS
C:\Users\a5484\Documents\Python\zhanggu\Lib\site-packages\numpy\core\getlimits.py:225: RuntimeWarning: invalid value encountered in exp2
  epsneg_f128 = exp2(ld(-113))
C:\Users\a5484\Documents\Python\zhanggu\Lib\site-packages\numpy\core\getlimits.py:226: RuntimeWarning: invalid value encountered in exp2
  tiny_f128 = exp2(ld(-16382))
C:\Users\a5484\Documents\Python\zhanggu\Lib\site-packages\numpy\core\getlimits.py:240: RuntimeWarning: invalid value encountered in exp2
  eps=exp2(ld(-112)),
C:\Users\a5484\Documents\Python\zhanggu\Lib\site-packages\numpy\core\getlimits.py:41: RuntimeWarning: invalid value encountered in nextafter
  self._smallest_subnormal = nextafter(
C:\Users\a5484\Documents\Python\zhanggu\Lib\site-packages\numpy\core\getlimits.py:52: RuntimeWarning: invalid value encountered in log10
  self.precision = int(-log10(self.eps))

Process finished with exit code -1073741819 (0xC0000005)
```

### Example data

no need to have a look data. error occurred during loading libraries. 

### Related existing issues or PRs

not found

ctuguinay, 2025-02-19:
Thanks for the issue, but from the error message this appears to be Numpy specific issue not an Echopype issue. Have you brought up this issue in the Numpy repository?

leewujung, 2025-02-21:
Yeah, and also @ZHANGGU you mentioned 
> Problem was due to Numpy version. Have tried upgrade Numpy to 2.1.3 but did not work either

But we do not support numpy>=2 yet, so you need to update your environment.

The easiest is to use conda, but pip install should work too. We're releasing v0.10.0 now and if you spin up a fresh environment and use pip install that should work.

ZHANGGU, 2025-02-21:
> Yeah, and also [@ZHANGGU](https://github.com/ZHANGGU) you mentioned
>
>
> But we do not support numpy>=2 yet, so you need to update your environment.
>
> The easiest is to use conda, but pip install should [...]

I have tried another PC, with venv installed Echopype 0.9.0. 
Numpy error is gone, but there are still problems. 
just simple codes as below

import os
from tqdm import tqdm
from echopype import open_raw

Then it pops up errors in loading the library

C:\Users\Administrator\Documents\MyPython\EchopypeVenv\Scripts\python.exe "C:/Program Files/JetBrains/PyCharm Community Edition 2023.1.1/plugins/python-ce/helpers/pydev/pydevd.py" --multiprocess --qt-support=auto --client 127.0.0.1 --port 59105 --file C:\Users\Administrator\PycharmProjects\Ad2cpConversion\convert_ad2cp.py 
Connected to pydev debugger (build 231.8770.66)
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "C:\Users\Administrator\Documents\MyPython\EchopypeVenv\Lib\site-packages\echopype\__init__.py", line 5, in <module>
    from . import calibrate, clean, commongrid, consolidate, mask, utils
  File "C:\Users\Administrator\Documents\MyPython\EchopypeVenv\Lib\site-packages\echopype\calibrate\__init__.py", line 1, in <module>
    from .api import compute_Sv, compute_TS
  File "C:\Users\Administrator\Documents\MyPython\EchopypeVenv\Lib\site-packages\echopype\calibrate\api.py", line 3, in <module>
    from ..echodata import EchoData
  File "C:\Users\Administrator\Documents\MyPython\EchopypeVenv\Lib\site-packages\echopype\echodata\__init__.py", line 7, in <module>
    from .echodata import EchoData
  File "C:\Users\Administrator\Documents\MyPython\EchopypeVenv\Lib\site-packages\echopype\echodata\echodata.py", line 12, in <module>
    from zarr.errors import GroupNotFoundError, PathNotFoundError
ImportError: cannot import name 'GroupNotFoundError' from 'zarr.errors' (C:\Users\Administrator\Documents\MyPython\EchopypeVenv\Lib\site-packages\zarr\errors.py)
python-BaseException

INFO: my virtual enviroment only has EchoPype, Tqdm. Nothing else.

leewujung, 2025-02-21:
@ZHANGGU : what you saw is a zarr>=3 problem, which we need more time to migrate to. In the upcoming v0.10.0 release echopype is partially upgraded and pinned to use all compatible version of various libraries. We still have the last thing to update, and will release as soon as that is done.

If you want to run with a new version now, please install from the `main` branch and use python=3.10-3.11 (**not** 3.12 - that is the last change we need to make for v0.10.0).

leewujung, 2025-02-23:
@ZHANGGU : v0.10.0 is now on pypi and conda, so I'll close this issue now. Feel free to reopen this or another issue if you run into other problems. 

Note that we are not actively developing the Nortek AD2CP part of the codebase due to time/funding constraint, so the code may not work with newer versions of data from these instruments. You are welcome to raise issues, which will give info for where the code support stands, or better yet, propose code changes through PRs!

closed 2025-02-23 by leewujung