echostack-org/echopype #1475: conflicting dimensions for 'ping_time' and 'ping_time': 'transmit_frequency_start'

gcutt, 2025-03-09:
### General description of problem

My EK80 files do not convert to echopype objects, using echopype 0.9.0 or 0.10.0. 

I use echopype's open_raw to try to convert to an echodata object, but it fails for all files. 

Here is the related code statement and error (full code and output are provided in the next sections). 
`
ed      = ep.open_raw( rawfile , sonar_model='EK80' )
`

Error: 
**ValueError: conflicting sizes for dimension 'ping_time': length 113 on 'ping_time' and length 50 on {'ping_time': 'transmit_frequency_start'}**

### Echopype version

echopype v0.10.0

### How did you install Echopype (e.g., conda or pip)

installed echopype using pip

### What is your operating system

windows

### Minimal code example

The following code reproduces the error: 

``
## imports 
import echopype as ep
print( f"echopype version: {ep.__version__}" )
import os

## Read raw EK80 data file to echopype object 
####  Following example from https://echopype.readthedocs.io/en/stable/convert.html
inpath  = r'example_data/test'
rawfn   = r'WBAT-EK80-TEST-0907-T235739.raw'
rawfile = os.path.join( inpath, rawfn )

ed      = ep.open_raw( rawfile , sonar_model='EK80' )

### Error message printouts

```shell
Full output: 
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[4], line 12
      8 rawfn   = r'WBAT-EK80-TEST-0907-T235739.raw'
     10 rawfile = os.path.join( inpath, rawfn )
---> 12 ed      = ep.open_raw( rawfile , sonar_model='EK80' )

File D:\ProgramData\Anaconda3\envs\echopype10\Lib\site-packages\echopype\utils\prov.py:237, in add_processing_level.<locals>.wrapper.<locals>.inner(*args, **kwargs)
    235 @functools.wraps(func)
    236 def inner(*args, **kwargs):
--> 237     dataobj = func(*args, **kwargs)
    238     if is_echodata:
    239         ed = dataobj

File D:\ProgramData\Anaconda3\envs\echopype10\Lib\site-packages\echopype\convert\api.py:516, in open_raw(raw_file, sonar_model, xml_path, include_bot, include_idx, convert_params, storage_options, use_swap, max_chunk_size)
    513 tree_dict["Sonar"] = None
    515 # Set multi beam groups
--> 516 beam_groups = setgrouper.set_beam()
    518 beam_group_type = []
    519 for idx, beam_group in enumerate(beam_groups, start=1):

File D:\ProgramData\Anaconda3\envs\echopype10\Lib\site-packages\echopype\convert\set_groups_ek80.py:1202, in SetGroupsEK80.set_beam(self)
   1200 for ch in self.sorted_channel["all"]:
   1201     if ch in self.sorted_channel["complex"]:
-> 1202         ds_data = self._assemble_ds_complex(ch)
   1203     elif ch in self.sorted_channel["power"]:
   1204         ds_data = self._assemble_ds_power(ch)

File D:\ProgramData\Anaconda3\envs\echopype10\Lib\site-packages\echopype\convert\set_groups_ek80.py:891, in SetGroupsEK80._assemble_ds_complex(self, ch)
    848 data_shape = self.parser_obj.ping_data_dict["complex"][ch]["real"].shape
    849 ds_tmp = xr.Dataset(
    850     {
    851         "backscatter_r": (
   (...)    888     },
    889 )
--> 891 ds_tmp = self._add_freq_start_end_ds(ds_tmp, ch)
    893 return set_time_encodings(ds_tmp)

File D:\ProgramData\Anaconda3\envs\echopype10\Lib\site-packages\echopype\convert\set_groups_ek80.py:817, in SetGroupsEK80._add_freq_start_end_ds(self, ds_tmp, ch)
    813     return ds_tmp
    815 print( f"L815*********************\nds_tmp \n{ds_tmp}\n*********************")
--> 817 ds_f_start_end = xr.Dataset(
    818     {
    819         "transmit_frequency_start": (
    820             ["ping_time"],
    821             freq_start.astype(float),
    822             self._varattrs["beam_var_default"]["transmit_frequency_start"],
    823         ),
    824         "transmit_frequency_stop": (
    825             ["ping_time"],
    826             freq_stop.astype(float),
    827             self._varattrs["beam_var_default"]["transmit_frequency_stop"],
    828         ),
    829     },
    830     coords={
    831         "ping_time": (
    832             ["ping_time"],
    833             self.parser_obj.ping_time[ch],
    834             self._varattrs["beam_coord_default"]["ping_time"],
    835         ),
    836     },
    837 )
    839 ds_tmp = xr.merge(
    840     [ds_tmp, ds_f_start_end], combine_attrs="override"
    841 )  # override keeps the Dataset attributes
    843 return ds_tmp

File D:\ProgramData\Anaconda3\envs\echopype10\Lib\site-packages\xarray\core\dataset.py:747, in Dataset.__init__(self, data_vars, coords, attrs)
    744 if isinstance(coords, Dataset):
    745     coords = coords._variables
--> 747 variables, coord_names, dims, indexes, _ = merge_data_and_coords(
    748     data_vars, coords
    749 )
    751 self._attrs = dict(attrs) if attrs else None
    752 self._close = None

File D:\ProgramData\Anaconda3\envs\echopype10\Lib\site-packages\xarray\core\dataset.py:460, in merge_data_and_coords(data_vars, coords)
    456     coords = create_coords_with_default_indexes(coords, data_vars)
    458 # exclude coords from alignment (all variables in a Coordinates object should
    459 # already be aligned together) and use coordinates' indexes to align data_vars
--> 460 return merge_core(
    461     [data_vars, coords],
    462     compat="broadcast_equals",
    463     join="outer",
    464     explicit_coords=tuple(coords),
    465     indexes=coords.xindexes,
    466     priority_arg=1,
    467     skip_align_args=[1],
    468 )

File D:\ProgramData\Anaconda3\envs\echopype10\Lib\site-packages\xarray\core\merge.py:705, in merge_core(objects, compat, join, combine_attrs, priority_arg, explicit_coords, indexes, fill_value, skip_align_args)
    700 prioritized = _get_priority_vars_and_indexes(aligned, priority_arg, compat=compat)
    701 variables, out_indexes = merge_collected(
    702     collected, prioritized, compat=compat, combine_attrs=combine_attrs
    703 )
--> 705 dims = calculate_dimensions(variables)
    707 coord_names, noncoord_names = determine_coords(coerced)
    708 if compat == "minimal":
    709     # coordinates may be dropped in merged results

File D:\ProgramData\Anaconda3\envs\echopype10\Lib\site-packages\xarray\core\variable.py:3022, in calculate_dimensions(variables)
   3020             last_used[dim] = k
   3021         elif dims[dim] != size:
-> 3022             raise ValueError(
   3023                 f"conflicting sizes for dimension {dim!r}: "
   3024                 f"length {size} on {k!r} and length {dims[dim]} on {last_used!r}"
   3025             )
   3026 return dims

ValueError: conflicting sizes for dimension 'ping_time': length 113 on 'ping_time' and length 50 on {'ping_time': 'transmit_frequency_start'}
```

### Example data

I can share a small file if someone can take a look at this problem. 

### Related existing issues or PRs

I have not seen similar issues reported.

leewujung, 2025-03-24:
Ugh, I just run into this myself! Thanks for reporting this @gcutt . I'll look into this second half of this next week.

gcutt, 2025-03-25:
Thank you! Let me know if there is anything I can do to help.
-Cutter

leewujung, 2025-04-07:
@gcutt : There turns out to be downstream changes that will take some time to address... Do you have files with different ping sequence patterns that you can provide as examples? I only have one set with 2 alternating patterns.

gcutt, 2025-04-07:
Hi, yes I can provide an example file tonight or tomorrow. Thanks

gcutt, 2025-04-12:
Hi Wu-Jung Lee,
This is George Cutter (github gcutt), contacting you about how to share a
EK80 file confidentially to help debug the echopype error (conflicting
dimensions for 'ping_time' and 'ping_time': 'transmit_frequency_start'
#1475).

Please let me know if there is a location that I can share a file (~20 to
100 MB size).

Thank you,
Cutter