echostack-org/echopype #1477: open_converted does not work with netcdf

seagrinch, 2025-03-13:
I'm having issues opening converted Netcdf files in echopype 0.10.

The following code works, if I simply go ahead and calibrate the EchoData object directly from `open_raw()`.  It also works if I use `open_converted()` using the .zarr file (first commented line).

```
ed = ep.open_raw(os.path.join(raw_file_dir,raw_file), sonar_model='EK80')
ed.to_zarr(save_path=converted_file, overwrite=True)
ed.to_netcdf(save_path=converted_file.replace('.zarr','.nc'), overwrite=True)

# ed = ep.open_converted(converted_file)
# ed = ep.open_converted(converted_file.replace('.zarr','.nc'))

# Calibrate the converted file
ds_Sv = ep.calibrate.compute_Sv(ed, waveform_mode="CW", encode_mode="complex") 
ds_Sv.to_zarr(calibrated_file, mode="w")
ds_Sv.to_netcdf(calibrated_file.replace('.zarr','.nc'), mode="w")

ds_Sv_new = ep.consolidate.add_location(ds_Sv, ed)
```

However, if I try to use the converted Netcdf file, I get the error...

> ValueError: dimensions ('channel',) must have the same length as the number of data dimensions, ndim=2

Interestingly, if I then try to rerun the code above, `to_netcdf()` throws an error.  (Bizarrely, this seems to only be a problem if you try to run the calibration function on the netcdf file.)

> KeyError: [<class 'netCDF4._netCDF4.Dataset'>, ('D20190119-T130524.nc',), 'a', (('clobber', True), ('diskless', False), ('format', 'NETCDF4'), ('persist', False)), '74a7988e-a9a5-412a-a914-bf2b9af92326']
>
> During handling of the above exception, another exception occurred:
> PermissionError: [Errno 13] Permission denied: './D20190119-T130524.nc'

I'm wondering if this may be related to the underlying netcdf engine issue noted in #988?

leewujung, 2025-03-24:
@seagrinch : Thanks for reporting this! Hmmm this is interesting... one of the main v0.10.0 changes was to get `open_converted` to work with the new xarray datatree feature, instead of the separate datatree package from before, but I don't think that would behave differently between zarr and netcdf file.

Could you provide a gist and a data file that I could try to reproduce the problem? I can look into this the second half of next week (BOAT workshop tomorrow, to a bit overwhelmed!)

seagrinch, 2025-07-16:
Hi @leewujung, apologies for the delay.  Here is a test file I was playing with...

https://marine.rutgers.edu/~sage/ek80_test/D20190119-T130524.raw

```
raw_file = 'D20190119-T130524.raw'
converted_file = raw_file.replace('.raw', '.zarr')
calibrated_file = raw_file.replace('.raw', '_Sv.zarr')

```

LOCEANlloydizard, 2025-08-08:
Hi @seagrinch @leewujung 

I've managed to reproduce the error, and after debugging I find the one line that caused the issue. The cause is in file [calibrate_ek.py](https://github.com/OSOceanAcoustics/echopype/blob/main/echopype/calibrate/calibrate_ek.py), in the CalibrateEK80 class, L548 ->

`tau_effective[ch_GPT] = beam["transmit_duration_nominal"][ch_GPT].isel(ping_time=0)`

replacing the line with

```
if ch_GPT.any():
       tau_effective[ch_GPT] = beam["transmit_duration_nominal"].sel(channel=ch_GPT).isel(ping_time=0)
```

removed the error. After that, I managed to run 

<pre>
import echopype as ep
import xarray as xr
import matplotlib.pyplot as plt

raw_file = "./D20190119-T130524.raw"
converted_zarr = raw_file.replace('.raw', '.zarr')
converted_nc = raw_file.replace('.raw', '.nc')

# Convert raw to Zarr and NetCDF
ed = ep.open_raw(raw_file, sonar_model='EK80')
ed.to_zarr(converted_zarr, overwrite=True)
ed.to_netcdf(converted_nc, overwrite=True)

# Open converted files
ed_zaar = ep.open_converted(converted_zarr)
ed_nc = ep.open_converted(converted_nc)

# Compute_Sv
ds_zaar = ep.calibrate.compute_Sv(ed_zaar, waveform_mode="CW", encode_mode="complex")
ds_nc = ep.calibrate.compute_Sv(ed_nc, waveform_mode="CW", encode_mode="complex")

# Plot both
fig, axes = plt.subplots(1, 2, figsize=(16, 6), sharey=True)

for ax, (ds, title) in zip(axes, [(ds_zaar, "Zarr"), (ds_nc, "NetCDF")]):
    ds["Sv"].plot(
        x="ping_time",
        ax=ax,
        vmin=-80,
        vmax=-30,
        cmap="RdYlBu_r",
        yincrease=False,
        add_colorbar=True
    )
    ax.set_title(title)

plt.tight_layout()
plt.show()

</pre>

and get the two plots

[image]

@seagrinch Note that this is a small fix you'd need to apply to your clone/fork for now if you want to continue processing the data
@leewujung Maybe instead of this fix, it would be better to first check why we have this difference between the Zarr and NetCDF files after open_converted()?

cheers!

BLayman-NOAA, 2025-08-08:
I ran into this issue as well. It seems to only happen with EK80 files from what I've seen. I created a repo to make it easier to reproduce if anyone is interested:

https://github.com/BLayman-NOAA/echopype-bug