echostack-org/echopype #1505: [Data conversion] Pulse length not in SV_OFFSET.

IanTBlack, 2025-06-04:
### General description of problem

I am trying to use echopype to process Ocean Networks Canada AZFP data.
I am running into an issue where echopype does not recognize the pulse length value that ONC sets for the 38kHz  (800 us) in some configurations.  I have confirmed that the configuration downloaded in the code example does set the pulse length for 38kHz to 800. Config snippet below.

<Frequency Number="1">
<AcquireFrequency units="bool">0</AcquireFrequency>
<PulseLen units="us">800</PulseLen>
<DigRate units="hz">20000</DigRate>
<RangeSamples units="samples">2811</RangeSamples>
<RangeAveragingSamples units="samples">3</RangeAveragingSamples>
<LockOutIndex units="samples">110</LockOutIndex>
<Gain units="gain">1</Gain>
<StorageFormat units="bool">0</StorageFormat>

Could this be resolved by adding an offset to the [SV_OFFSET dictionary](https://github.com/OSOceanAcoustics/echopype/blob/358736f11de02330fdf09731019100f16bf79ada/echopype/convert/parse_azfp.py#L28)? I am not a regular user of AZFP data, so I am unsure of what the offset value would be.

Thanks!

### Echopype version

0.10.1

### How did you install Echopype (e.g., conda or pip)

pip install echopype

### What is your operating system

Windows 11, Python 3.10

### Minimal code example

This code downloads example ONC AZFP data. You will need to sign up for an account and obtain a token from [Ocean Networks Canada](https://data.oceannetworks.ca/Login).  

```
from datetime import datetime
from netrc import netrc

from onc import ONC

_, __ , ONC_TOKEN = netrc().authenticators(host = 'data.oceannetworks.ca')
onc = ONC(token = ONC_TOKEN)

BDT = datetime(2025, 5, 15, 1 ,0, 0)  # Request begin datetime.
EDT = datetime(2025, 5, 15, 1, 59, 59, 999999) # Request end datetime.

req_dp_params = {
    'token': ONC_TOKEN,
    'locationCode': 'PVIP',
    'deviceCategoryCode': 'ECHOSOUNDERBIOA',
    'dataProductCode': 'AAPTS', 
    'extension': '01a',
    'dateFrom': BDT.strftime('%Y-%m-%dT%H:%M:%S.%f')[:-3] + 'Z',
    'dateTo': EDT.strftime('%Y-%m-%dT%H:%M:%S.%f')[:-3] + 'Z',
    'dpo_aslBinarySource': 0
}

dpreq = onc.requestDataProduct(filters = req_dp_params)
dpstatus = onc.checkDataProduct(dpreq['dpRequestId'])
dprun = onc.runDataProduct(dpreq['dpRequestId'])
download = onc.downloadDataProduct(dprun['runIds'][0], overwrite = True)
```

Then this is the echopype code I used to open the downloaded file.
```
import echopype as ep

raw_path = 'output/ASLAZFP55167_20250515T010000.235Z_20250515T015958.172Z.01a'
xml_path = 'output/ASLAZFP55167_20250515T010000.235_20250515T015958.172Z.xml'

ed = ep.open_raw(raw_path, sonar_model='AZFP', xml_path = xml_path)
```

### Error message printouts

```shell
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[10], line 3
      1 raw_path = 'output/ASLAZFP55167_20250515T010000.235Z_20250515T015958.172Z.01a'
      2 xml_path = 'output/ASLAZFP55167_20250515T010000.235_20250515T015958.172Z.xml'
----> 3 ed = ep.open_raw(raw_path, sonar_model='AZFP',xml_path = xml_path)

File ~\phd\projects\ONCHackathon\.venv\lib\site-packages\echopype\utils\prov.py:237, in add_processing_level.<locals>.wrapper.<locals>.inner(*args, **kwargs)
    235 @functools.wraps(func)
    236 def inner(*args, **kwargs):
--> 237     dataobj = func(*args, **kwargs)
    238     if is_echodata:
    239         ed = dataobj

File ~\phd\projects\ONCHackathon\.venv\lib\site-packages\echopype\convert\api.py:472, in open_raw(raw_file, sonar_model, xml_path, include_bot, include_idx, convert_params, storage_options, use_swap, max_chunk_size)
    461 parser = SONAR_MODELS[sonar_model]["parser"](
    462     file_chk,
    463     # Currently used only for AZFP XML File
   (...)
    469     sonar_model=sonar_model,
    470 )
    471 # Actually parse the raw datagrams from source file
--> 472 parser.parse_raw()
    474 # Direct offload to zarr and rectangularization only available for some sonar models
    475 # No rectangularization for other sonar models not listed below
    476 if sonar_model in ["EK60", "ES70", "EK80", "ES80", "EA640"]:
    477     # Perform rectangularization and offload to zarr
    478     # if the data expansion is too large to fit in memory

File ~\phd\projects\ONCHackathon\.venv\lib\site-packages\echopype\convert\parse_azfp.py:381, in ParseAZFP.parse_raw(self)
    379 self.Sv_offset = np.zeros_like(self.freq_sorted)
    380 for ind, ich in enumerate(self.freq_ind_sorted):
--> 381     self.Sv_offset[ind] = self._calc_Sv_offset(
    382         self.freq_sorted[ind], self.unpacked_data["pulse_len"][ich]
    383     )

File ~\phd\projects\ONCHackathon\.venv\lib\site-packages\echopype\convert\parse_azfp.py:575, in ParseAZFP._calc_Sv_offset(freq, pulse_len)
    573 # Check if the specified freq-pulse length combination is in the allowable Sv_offset dict
    574 if pulse_len not in SV_OFFSET[freq]:
--> 575     raise ValueError(
    576         f"Pulse length {pulse_len} us is not in the Sv offset dictionary "
    577         f"for the {freq} Hz channel! "  # add freq info
    578         "Please contact AZFP Environmental Sciences "
    579         "and raise an issue in the echopype repository."
    580     )
    582 return SV_OFFSET[freq][pulse_len]

ValueError: Pulse length 800 us is not in the Sv offset dictionary for the 38000.0 Hz channel! Please contact AZFP Environmental Sciences and raise an issue in the echopype repository.
```

### Example data

Example to download data in code example. Can send .01a and .xml if needed.

### Related existing issues or PRs

NA

jchawarski, 2025-12-05:
Hi Ian, 

Julek here from ASL. I'm starting to work with echopype and running into similar problems when reading AZFP data with pulse lengths that don't match the Sv offset dictionary. We are going to implement a solution that provides an update to the dictionary so that pulse lengths between values are linearly interpolated or use nearest neighbor. For your reference, these offsets are stored in the attached operators manual pdf Table 3, printed here:
 
[image]

and just a different representation of that table, where we'll interpolate values between those listed so that users don't run into this error and all offsets are applied appropriately without user needing to intervene:

[image]

[GU-100-AZFP-01-R94 Operator's Manual.pdf](https://github.com/user-attachments/files/23971926/GU-100-AZFP-01-R94.Operator.s.Manual.pdf)

IanTBlack, 2025-12-05:
Awesome. Thanks for the info. 
My temporary workaround without having to modify the echopype package source has been to manually set the Sv offset.
I've been using 0.9 as a guess, which I think would fit within the table you provided. 

```
import echopype
echopype.convert.parse_azfp.SV_OFFSET[38000.0][800] = 0.9
```

Looking forward to the interpolation update. I'll be sure to reprocess once that goes through.

leewujung, 2025-12-05:
@IanTBlack : Sorry for the lack of response - I think your issue came while I was getting ready for the field, and also that we didn't have the matched values to use before. Glad that you were able to find a workaround.

@jchawarski : Looking forward to the PR that can solve this problem once and for all (or most...)! We've seen multiple issues on this and have run into this ourselves as well.

IanTBlack, 2025-12-06:
@jchawarski  I took a crack at an interpolation function based on Table 3. Wasn't sure how to handle pulse lengths outside 150 and 1000 us (or if such applications exist) so I have it raise a ValueError in those instances.

```
import numpy as np

def calculate_azfp_sv_offset(pulse_length: float | int, 
                          frequency: float | int) -> float:
    """
    Linearly interpolate an SV offset for a given pulse length and frequency.
    The predefined offsets used in this function can be found in Table 3 of the AZFP Operator's Manual.

    :param pulse_length: A pulse length value (in microseconds) from an AZFP config file for a given frequency channel.
    :param frequency: The AZFP frequency channel.
    :return: Either the known Sv offset if it is predefined, or an interpolated value for less common pulse lengths.
    """
    pulse_length = int(pulse_length) # Convert to an integer for consistency.
    frequency = int(frequency) 
    
    high_frequencies = [120000,200000,455000,769000] # AZFP high frequency channels.
    sv_offset_hf = {  # Known high frequency channel correction values.
        150:1.4,
        200:1.4,
        250:1.3,
        300:1.2,
        500:0.9,
        700:0.6,
        900:0.3,
        1000:0.2
    }
    
    low_frequencies = [38000,67500] # AZFP low frequency channels.
    sv_offset_lf = { # Known low frequency channel correction values.
        500:1.1,
        1000:0.7
    }

    if pulse_length < 150 or pulse_length > 1000: 
        raise ValueError("Pulse lengths less than 150 or greater than 1000 microseconds are not supported.")
    
    if frequency in high_frequencies:
        known_offsets = sv_offset_hf
    elif frequency in low_frequencies:
        if pulse_length < 500:
            raise ValueError("Pulse lengths less than 500 microseconds for low frequencies are not supported.")
        known_offsets = sv_offset_lf
        

    if pulse_length in known_offsets.keys(): # If the pulse length already has a known offset, just return that without interpolating.
        known_offset = known_offsets[pulse_length]
        return known_offset
    else: # If the pulse length doesn't have a known offset, linearly interpolate.
        (x,y) = zip(*known_offsets.items())
        interpolated_offset = np.interp(pulse_length, x, y)
        return interpolated_offset
```

which when used for a pulse length of 800 us at 38kHz gives 0.86.

```
pulse_length = 800
frequency = 38000.0

offset = calculate_azfp_sv_offset(pulse_length, frequency)
```

leewujung, 2025-12-09:
This is the way 🤖